Timendus/ticalc-usb #5: Not working for TI-84 Plus CE-T

PrivateDragon, 2020-09-05:
I have tried if it worked on a linux pc for the TI-84 Plus CE-T, because the TI-Connect software is not available for Linux.
But it didn't work.
So: On Linux, it doesn't work for the TI-84 Plus CE-T

adriweb, 2020-09-05:
Yes, the version used on the website currently doesn't have all the improvements made on the source code yet. Some PRs need to be merged etc first.

PrivateDragon, 2020-09-05:
I don't know if the problem is either in the website or because I use Linux. 
My laptop didn't recognize the calculator as a device...
I couldn't find it.
So maybe it's a problem with the communication between Linux and the TI-84 Plus CE-T?

adriweb, 2020-09-05:
No that's the thing - the current version is only going to work with the TI-84 Plus calculator, not the CE just yet. For that you'd have to "build" the latest sources (well from the PR)

PrivateDragon, 2020-09-06:
Is anyone going to do that in the near future?

Jeffitus, 2020-09-06:
Probably yes eventually; since you're on Linux I recommend installing TiLP. You can find info on installing it here: http://lpg.ticalc.org/prj_tilp/linux.html

adriweb, 2020-09-06:
(Or rather, for the latest version of the install script, here: https://raw.githubusercontent.com/debrouxl/tilp_and_gfm/master/tilp/trunk/build/scripts/install_tilp.sh)

PrivateDragon, 2020-09-06:
The problem with TiLP is that I have to ask my father to install it on my laptop, and when he looked at the dependencies he said that it might not work on my laptop because I have KDE instead of something else..?
And he doesn't really know how to get the latest version from Github.
(The version on github is very recently updated I think)

adriweb, 2020-09-06:
In the link I gave you, it has the dependencies listed. That will get&build the latest version from github.

debrouxl, 2020-09-06:
Timendus' readme for this repo lays out the steps you need to take to build it. I used them several weeks ago.

TILP is desktop-agnostic. It used to have an integration with KDE 3 file dialogs, but I removed that a while ago. I know that there's Trinity, but that's even far more of a dwarf in terms of real-world usage than KDE 5...

closed 2020-11-29 by Timendus