Timendus/ticalc-usb #31: Calculator support request

aPerson000, 2021-09-22:
## Device information

{
  "deviceClass": 0,
  "deviceProtocol": 0,
  "deviceSubclass": 0,
  "deviceVersionMajor": 2,
  "deviceVersionMinor": 4,
  "deviceVersionSubminor": 0,
  "manufacturerName": "Texas Instruments Incorporated",
  "productId": 57360,
  "productName": "TI-84 Plus CE",
  "serialNumber": "13008FA281042D0E",
  "usbVersionMajor": 2,
  "usbVersionMinor": 0,
  "usbVersionSubminor": 0,
  "vendorId": 1105
}

## Additional information

Please provide this extra information so we can better support your calculator:

Calculator name as written on the device: TI-84 Plus CE
This calculator can also receive files made for these other types of calculators: TI-83 Premium CE, TI-84 Plus
Any other technical information that might help: ...

## Please note

We do not promise to support any calculators, even if there are a lot of requests for them. We're just hobbyists with limited time available. There may be technical reasons why one device is easier to support than another. We may not have access to the device and not be able to test it. So please don't be offended if your request goes unanswered.

Timendus, 2021-09-22:
Hello @aPerson000, thank you for your support request! We have just bumped TI-84 Plus CE support to "beta" status, and it should now mostly just work. Were you sent here from ticalc.link?

Timendus, 2021-11-18:
Hey @aPerson000, I'm going to close this issue since this device should be supported and you didn't answer my question :) Feel free to open another one if you feel this was in error!

closed 2021-11-18 by Timendus