Timendus/ticalc-usb #33: Archiving files

reesericci, 2021-09-27:
On ticalc.link, there is currently no way to archive files when transferring like in TI-Connect. Is there any way to add this?

--reese

Timendus, 2021-09-30:
Do you mean that when transferring the file ends up on the calculator, already archived? I guess that would be a matter of setting the right property, but I'd have to figure that out and build it into ticalc-usb. And seeing that you can achieve the same result by archiving the file on the device after the transfer, it's not really a priority at the moment.

(I wrote ticalc.link and ticalc-usb because I didn't find any linking software that "just worked" and that I liked for MacOS, after about a decade of not touching TI calculators. So ironically enough, I'm really not familiar with the features of other software 🙈 Please let me know if you meant something completely different!)

reesericci, 2021-09-30:
> Do you mean that when transferring the file ends up on the calculator, already archived?
yes! TI-Connect can do this, and it would be helpful to have in ticalc.link

Timendus, 2021-09-30:
Alright, I'm sure someone reading this will be able to confirm if I'm correct in thinking that this would be a matter of setting `DUSB_AID_ARCHIVED` to true instead of the value of the `archived` attribute of the file in these lines:

https://github.com/Timendus/ticalc-usb/blob/f5b30333333bd730bd0ee56d232bcadbcb5c4951/src/dusb/ti84series.js#L125-L129

If that's the case, it wouldn't be much work, I guess.

reesericci, 2021-09-30:
@debrouxl ^ ?

debrouxl, 2021-09-30:
Yup. Simply add a mode where you force entry.attributes.archived to true before passing said entry to the variable sending code.