Timendus/ticalc-usb #34: Calculator support request

SuperX-dev, 2021-09-29:
## Device information
{
  "deviceClass": 0,
  "deviceProtocol": 0,
  "deviceSubclass": 0,
  "deviceVersionMajor": 1,
  "deviceVersionMinor": 0,
  "deviceVersionSubminor": 5,
  "manufacturerName": "Texas Instruments Incorporated",
  "productId": 57362,
  "productName": "TI-Nspire(tm) Handheld",
  "serialNumber": null,
  "usbVersionMajor": 2,
  "usbVersionMinor": 0,
  "usbVersionSubminor": 0,
  "vendorId": 1105
}
## Additional information

Please provide this extra information so we can better support your calculator:

Calculator name as written on the device: Ti Nspire CX (Hardware Revision Y in this case)
This calculator can also receive files made for these other types of calculators: Other Ti Nspire Cx´s and Ti nspire CX II (Don´t have others to test it)
Any other technical information that might help: Tilp works with this calculator even though it has some bugs

##My message:
Really looking forward to see this project growing. I found this because of an issue I opened [https://github.com/debrouxl/tilp_and_gfm/issues/47](url), because it would be really useful to send files from my phone to the calculator, no more need to always have to carry my laptop if I want to do that, just use the OTG adapter in my phone. 
And actually my OnePlus Nord using Google Chrome recognises it and everything went through just like the PC version, which really gets me hyped to see the potential that this project can achieve.

Here is a screenshot of my phone detecting it:
[image]

Pretty great and keep up with you great work

debrouxl, 2021-09-29:
You can use https://lights0123.com/n-link/ to communicate with the Nspire family, including the CX II. The libnspire N-Link is based on has narrow NavNet & NNSE protocol support, and cuts down on too many layers), but by now, after enough fixes by Vogtinator, it does its one job - file-related operations - quite reliably, I'm told.
Unlike libticalcs, it supports the CX II series' NNSE protocol.

SuperX-dev, 2021-09-29:
> You can use https://lights0123.com/n-link/ to communicate with the Nspire family, including the CX II. The libnspire N-Link is based on has narrow NavNet & NNSE protocol support, and cuts down on too many layers (*), but by now, after enough fixes by Vogtinator, it does its one job - file-related operations - quite reliably, I'm told :) Unlike libticalcs, it supports the CX II series' NNSE protocol, which is some form of wrapping on top of a variant of the NavNet protocol used by older Nspire models.
> 
> libti*/tilp have layers (and not even enough of them yet) because they make it possible to communicate with the whole set of TI graphing calculators through 4 different protocols using a unified API, instead of reinventing similar wheels across different projects.

Wow this is just amazing. The Chrome version worked on my phone like a charm. I was searching for this for so long and now it came true. Thank you, now my friends and I don´t have to carry around a laptop to just transfer some games to our calculators or classes things. Thank You for your great work all of you. ;)

closed 2021-09-29 by SuperX-dev

debrouxl, 2021-09-29:
Well, it's still desirable for ticalc-usb+ticalc.link to support multiple calculator families - it can only be a good thing for the code base :)

Timendus, 2021-09-30:
Hey there @SuperX-dev, thanks for making (and closing) an issue! I know that communicating with calculators through WebUSB also works on phones and tablets (at least on Android), which is one of its many promising qualities, and cross-platform compatibility is one of the reasons to start this project. Good to see you found a nice alternative 😄