Timendus/ticalc-usb #38: Calculator support request (TI-84 Plus CE)

Bovemax, 2021-12-27:
{
  "deviceClass": 0,
  "deviceProtocol": 0,
  "deviceSubclass": 0,
  "deviceVersionMajor": 2,
  "deviceVersionMinor": 4,
  "deviceVersionSubminor": 0,
  "manufacturerName": "Texas Instruments Incorporated",
  "productId": 57360,
  "productName": "TI-84 Plus CE",
  "serialNumber": "1300BE3FEF042D0E",
  "usbVersionMajor": 2,
  "usbVersionMinor": 0,
  "usbVersionSubminor": 0,
  "vendorId": 1105
}

Timendus, 2022-02-09:
Hello there @Bovemax, sorry for the late reply. Your issue surprised me a bit, because `ticalc-usb` already supports the TI-84 Plus CE (at support level beta, but that's pretty much as good as it gets right now).

However, I see that your device reports itself as `productID: 57360`, whereas the TI-84 Plus CE should have `productID: 57352`. That makes it so that the library doesn't recognize your device as a TI-84 Plus CE.

Did you try again to see if this was a fluke? Does your device work with other linking software?

Timendus, 2022-02-09:
According to [this](https://www.the-sz.com/products/usbid/index.php?v=0x0451&p=0xe010&n=) a device with your ID should be a "TI SmartPad Keyboard" 🙈

adriweb, 2022-02-09:
Oh maybe the CE was running the TI SmartPad app (thus behaving as a keyboard, basically).
Anyway, should be fine once the calc is back in "normal" mode.

Timendus, 2022-02-09:
Ah! I didn't know that was a thing. Good to know.

closed 2022-04-15 by Timendus