Timendus/ticalc-usb #39: Calculator support request

jasonm5065, 2022-01-27:
## Device information

[Please copy-and-paste the output that ticalc.link gives you for your device here]

## Additional information

Please provide this extra information so we can better support your calculator:

Calculator name as written on the device: ...
This calculator can also receive files made for these other types of calculators: ...
Any other technical information that might help: ...

## Please note

We do not promise to support any calculators, even if there are a lot of requests for them. We're just hobbyists with limited time available. There may be technical reasons why one device is easier to support than another. We may not have access to the device and not be able to test it. So please don't be offended if your request goes unanswered.

commandblockguy, 2022-01-27:
Unfortunately, we can't read minds (quite yet, anyways) - you'll need to fill out the device information fields for us to know what calculator you want us to add support for.

As an aside, this is the second time that someone has submitted this template without filling any information - I wonder if people are misinterpreting the name of the template as "Click here to request someone to provide you with support for using the library/website," as in "support ticket," rather than "Click here to request that support be added to the library for a new type of calculator," as in "supported feature?" It might be worth rephrasing the template to clarify this point.

Timendus, 2022-02-09:
Ah! Good point @commandblockguy. That could be. Or maybe people just don't really get what they're doing on Github.

Timendus, 2022-02-09:
If people are coming through the ticalc.link website, this is the interaction they should see:

```
Sorry!
It looks like your device is not yet supported. Would you like to submit it for consideration?
```
When answering "yes":
```
## Device info to submit ##

Please [file a support request on Github] with the following information:

{
      deviceClass: device.deviceClass,
      deviceProtocol: device.deviceProtocol,
      [etc]
}
```
And the link to Github should lead them to the above template.

closed 2022-04-15 by Timendus