Timendus/ticalc-usb #60: Calculator support request

packrunnerlucas, 2024-05-02:
## Device information

{
  "deviceClass": 0,
  "deviceProtocol": 0,
  "deviceSubclass": 0,
  "deviceVersionMajor": 2,
  "deviceVersionMinor": 2,
  "deviceVersionSubminor": 0,
  "manufacturerName": "Texas Instruments Incorporated",
  "productId": 57352,
  "productName": "SerCertNotSet",
  "serialNumber": null,
  "usbVersionMajor": 2,
  "usbVersionMinor": 0,
  "usbVersionSubminor": 0,
  "vendorId": 1105
}

## Additional information

Please provide this extra information so we can better support your calculator:

Calculator name as written on the device: ..TI-84 Plus CE
This calculator can also receive files made for these other types of calculators: ...
Any other technical information that might help: ...

## Please note

We do not promise to support any calculators, even if there are a lot of requests for them. We're just hobbyists with limited time available. There may be technical reasons why one device is easier to support than another. We may not have access to the device and not be able to test it. So please don't be offended if your request goes unanswered.

debrouxl, 2024-05-04:
Interesting, this looks like a 84+CE with destroyed certificate memory contents, like one of my two 83PCE.

packrunnerlucas, 2024-05-04:
Idk I jailbroke it with a few different applications. Everything works as
intended though. If it doesn’t work no biggie I can just use hdpic
conversion

debrouxl, 2024-05-04:
AFAICT, it's not the jailbreaks. I broke my calculator's certificate memory simply while transferring FlashApps as part of libti* tests. Possibly when interrupting a transfer.

packrunnerlucas, 2024-05-04:
I’m not very fluent in the calculator language. Is it a cause for concern
or just a cool tidbit?

debrouxl, 2024-05-04:
For the most part, your and my calculator, which aren't the only SerCertNotSet NoCertProductName calculators in the wild, still work. Sure, some tools had to / will have to implement minor workarounds for this peculiarity, but I'm not currently aware of induced misbehaviour which would make this peculiarity more of a cause for concern than a cool tidbit :)

Not that I tried to transfer any more FlashApps to my 83PCE which identifies itself as SerCertNotSet NoCertProductName after noticing the change, though; I got a 83PCEEP and switched to it.

adriweb, 2024-05-04:
It should probably be fixable with CerMastr, though. And thus restore functionality with tools that depend on the name.
But there's no ready-to-use tutorials for this...

packrunnerlucas, 2024-05-05:
Ah it’s fine don’t worry about it brother. Thanks for the pointers