Timendus/ticalc-usb #63: Calculator support request

theevilsnips, 2024-07-19:
## Device information

{
  "deviceClass": 0,
  "deviceProtocol": 0,
  "deviceSubclass": 0,
  "deviceVersionMajor": 2,
  "deviceVersionMinor": 4,
  "deviceVersionSubminor": 0,
  "manufacturerName": "Texas Instruments Incorporated",
  "productId": 57360,
  "productName": "TI-84 Plus CE",
  "serialNumber": "1301009C59042D0E",
  "usbVersionMajor": 2,
  "usbVersionMinor": 0,
  "usbVersionSubminor": 0,
  "vendorId": 1105
}

## Additional information

Please provide this extra information so we can better support your calculator:

Calculator name as written on the device: ... TI-84 Plus CE PYTHON
This calculator can also receive files made for these other types of calculators: ...
Any other technical information that might help: ...

## Please note

We do not promise to support any calculators, even if there are a lot of requests for them. We're just hobbyists with limited time available. There may be technical reasons why one device is easier to support than another. We may not have access to the device and not be able to test it. So please don't be offended if your request goes unanswered.

ArrowlTCS, 2024-09-13:
AFAIK, this calculator is already supported, right?

adriweb, 2025-09-18:
Hm this should already work.
Does it work with https://ticonnectce.latealways.dev/ ?

closed 2025-09-18 by adriweb