Timendus/ticalc-usb #65: Calculator support request

5ghzx, 2024-09-23:
## Device information

{
  "deviceClass": 0,
  "deviceProtocol": 0,
  "deviceSubclass": 0,
  "deviceVersionMajor": 2,
  "deviceVersionMinor": 2,
  "deviceVersionSubminor": 0,
  "manufacturerName": "Texas Instruments Incorporated",
  "productId": 57352,
  "productName": "\u0004I-84 Plus CE",
  "serialNumber": null,
  "usbVersionMajor": 2,
  "usbVersionMinor": 0,
  "usbVersionSubminor": 0,
  "vendorId": 1105
}

## Additional information

It's a TI-84 plus CE, but the name seems to be corrupted, so it doesn't work. It did work before I shut it down while an app was installing. After it only showed a white screen, so i reinstalled the OS.

## Please note

We do not promise to support any calculators, even if there are a lot of requests for them. We're just hobbyists with limited time available. There may be technical reasons why one device is easier to support than another. We may not have access to the device and not be able to test it. So please don't be offended if your request goes unanswered.

adriweb, 2024-09-23:
(You should be able to fix it using [CerMastr](https://tiplanet.org/forum/archives_voir.php?id=1581757)

5ghzx, 2024-09-24:
I checked, and the name shows up normally in ti connect. Does that change anything?

adriweb, 2024-09-24:
Ti connect doesn't really care about the name. Unfortunately ticalc.link detects the calc through that, among other things.